libretiny-eu/upk2esphome #2: LSC - 3004200 WiFi Outdoor Dual Socket *Working Config*

Sn00kiT, 2023-05-07:
Working Config for the LSC Dual Socket Outdoor Plug:

```yml
button:
- platform: restart
  name: Restart

debug:
  update_interval: 30s

switch:
- platform: gpio
  pin: P14
  name: "Relay #1 On"
  id: relay1
  interlock: [relay2]
#  internal: true
- platform: gpio
  pin: P26
  name: "Relay #2 Off"
  id: relay2
  interlock: [relay1]
#  internal: true

- platform: gpio
  pin: P9
  name: "Relay #3 On"
  id: relay3
  interlock: [relay4]
#  internal: true
- platform: gpio
  pin: P8
  name: "Relay #4 Off"
  id: relay4
  interlock: [relay3]
#  internal: true

- platform: template
  name: "Switch_Socket1"
  id: Socket_1
  optimistic : true
  turn_on_action:
    - switch.turn_off: relay2
    - switch.turn_on: relay1
  turn_off_action:
    - switch.turn_off: relay1
    - switch.turn_on: relay2
    
- platform: template
  name: "Switch_Socket2"
  id: Socket_2
  optimistic : true
  turn_on_action:
    - switch.turn_off: relay4
    - switch.turn_on: relay3
  turn_off_action:
    - switch.turn_off: relay3
    - switch.turn_on: relay4

text_sensor:
- platform: debug
  reset_reason:
    name: Reset Reason
- platform: libretiny
  version:
    name: LibreTiny Version

sensor:
- platform: uptime
  name: Uptime

binary_sensor:
- platform: gpio
  id: binary_switch_all
  pin:
    number: P7
    inverted: true
  on_press:
    then:
    - switch.toggle: Socket_1
    - switch.toggle: Socket_2

status_led:
  pin: P1
```

kuba2k2, 2023-05-13:
Is it different than the one generated by UPK? Did the UPK config have any problems?

Sn00kiT, 2023-05-16:
> Is it different than the one generated by UPK? Did the UPK config have any problems?

They YAML generated in UPK does only have the basic config and no GPIO´s associated. When using this the device simply does not have any functions.