libretiny-eu/upk2esphome #5: device support

bkbartk, 2023-09-20:
Hello,

I don't know it this is the place to ask, I was looking for some kind of Q&A page or message board but couldn't find any.
Are there plans to extend the device support to other devices as well?

there are 2 devices which I want to OTA flash, but I'm a little bit afraid I will kill the functionality if I don't know the config.
I'm especially interested in vacuum cleaner support. this makes use of datapoints over tx/rx instead of GPIO pins. So I don't even know if this is possible,

but the main question is, do you have plans to support more device types if possible?

kuba2k2, 2023-09-20:
What other devices specifically do you mean?

The vacuum cleaner, does it have BK7231? These usually have something more powerful, even Linux sometimes.

"datapoints over tx/rx" is simply TuyaMCU. UPK2ESPHome supports TuyaMCU (partially), so you can try that.

bkbartk, 2023-09-20:
> What other devices specifically do you mean?

specifically I mean vacuum cleaners, Also doorbells, but those mostly run linux busybox, so It might be another challenge.

> The vacuum cleaner, does it have BK7231? These usually have something more powerful, even Linux sometimes.

I actually don't know for sure, I was expecting it to be Either BK7231 or ESP8266 since my model is from 2021.
Is there a way to know this without opening the device?
I the tuya app the firmware says:
```
Main Module: v2.1.6
MCU Module: v6.0.1
```
not sure if it's possible to derive the chipinformation from here, Or maybe I should just try to use cloudcutter and see if it works.

> "datapoints over tx/rx" is simply TuyaMCU. UPK2ESPHome supports TuyaMCU (partially), so you can try that.

true, but if it doesn't work, my vaccuum cleaner will probably brick,
however, I found this one, 
https://github.com/Rjevski/esphome-eufy-robovac-g10-hybrid
So I could give this approach a try if UPK2ESPHome  won't work, But I was hoping it would be easier to have something autogenerated to avoid mistakes.

daniel-dona, 2023-09-21:
@bkbartk maybe you can check if the vacuum cleaner is here listed https://robotinfo.dev/, if so it's possible that you can use Valetudo https://github.com/Hypfer/Valetudo 

What's the manufacturer / model?

bkbartk, 2023-09-21:
My Vucuum cleaner is not listed here,
It's a "Medion S20 SW" (Purchased in The Netherlands)
I can try this but to have it off cloud I use "Local Tuya" and block all internet traffic from the device. The only thing not provided here is cleaning records.

I tried Cloudcutter but it doesn't work, so the chip probably is not Beken.

daniel-dona, 2023-09-21:
You can also try to get the vendor name using the MAC of the device and some web pages like this https://www.macvendorlookup.com/

If you're lucky, you'll see something like "Beken Corporation", if not only "Tuya Smart Inc."

bkbartk, 2023-09-21:
unfortunately I'm not lucky
[image]
however, I do see Tuya is an American company

bkbartk, 2023-09-22:
this gave me an idea,
I checked a tuya device with an ESP8266 chip inside,
it says:
`Espressif Inc.`
And one with a beken chip inside,
it says: 

```
Company
    Tuya Smart Inc.
Address
    FLAT/RM 806 BLK ? 8/F CHEUNG SHA WAN PLAZA 833 CHEUNG SHA WAN ROAD KL
    hongkong Hong Kong 999077
    CN
Range
    70:89:76:00:00:00 - 70:89:76:FF:FF:FF
Type
    IEEE MA-L 
```

also tuya, but a different address and range,
so at this point I can rule out that an espchip is inside.