libretiny-eu/upk2esphome #7: WB3S Tuya 1 gang switch

alecsandes, 2023-11-23:


alecsandes, 2023-11-23:
I used the 3 gang version, and in relay 1, toggling on nothing happens, toggling off, the relay activates and the toggle as well.
When flashing via tuya cloud cutter, I have chosen the profile:
devices/lonsonho-x801a-1-gang-switch.json instead of Tuya generic ATS01L-US 1 Gang Touch Switch

Cossid, 2023-11-24:
For unlisted devices, your best bet is installing ESPHome-Kickstart and using the upk2esphome plugin portion of LTChipTool to scan your kickstart device to see if it has device configuration in storage, the ltchiptool will generate a similar profile if the data is there.  Otherwise you can use the included pinscan to test pins.  There is not more specific advice we can give.

If a relay is backwards, you need to invert the pin.

alecsandes, 2023-11-24:
Wow! I am so surprised to have the reply so quick!

For me it is unclear, if I start the tuya-cloudcutter program and choose a device by chip type and fw version (like I did (choose in the begining devices/lonsonho-x801a-1-gang-switch in the list, but then choosen the ESPHomeKickstarter, won't it be having the configuration of what I have chosen when flashing?

Now, I am back on ESPHome kickstart fw. will try now the LTChiptool.

I have other questions as well:
1. is there a way to go back to tuya fw?
2. once flashed with ESPHome, I cannot dump anymore the tuya configiuration via serial?
3. If I have an untouched device, can i use the LTChipTool to read the tuya configuration only by soldering TX/RX pins or I need to remove some chip like in case of FW update via serial?
4. I have seen that the status lights with the Tuya Generic ATS01L-US 1 gang Touch switch are fading on interaction, which was not before, how can I do replicate it by yaml code building in LTChipTool?

TIA!

kuba2k2, 2023-11-24:
> For me it is unclear, if I start the tuya-cloudcutter program and choose a device by chip type and fw version (like I did (choose in the begining devices/lonsonho-x801a-1-gang-switch in the list, [...]

tuya-cloudcutter allows you to flash custom firmware, like ESPHome. It doesn't change any GPIO/pin configuration. What it does though, is change the device's secret keys, so that you cannot connect it back to Tuya Cloud.

---

> Now, I am back on ESPHome kickstart fw. will try now the LTChiptool.

That is the correct way to go. ltchiptool will allow you to pull device config from the switch on runtime (as long as Kickstart is installed) and will generate a (mostly) ready-to-use ESPHome YAML with correct pin config (as long as it's possible - should be).

---

1. No, unless you have dumped the firmware (via serial) before using tuya-cloudcutter. If you have, you can simply reflash the dumped firmware via serial again.
2. You can. Tuya-cloudcutter doesn't overwrite the configuration (apart from device secrets). However, dumping via serial will give you exactly the same result as via ltchiptool+UPK2ESPHome.
3. I don't know what chip you had to remove, I don't see any removed chip on the photo. "Tuya configuration" is stored in flash, same as firmware. So the same steps apply here like in case of FW update via serial.
4. They shouldn't be fading, the YAML you have is probably wrong. Generate YAML with UPK2ESPHome and try again.

alecsandes, 2023-12-09:
I come back with feedback, the kickstarter is great! It made the yaml just right reading from device.
The challenge is to find the right profile when flashing the device.

Well, almost everything is great except:
1. Multigang switches
Multigang switches after flashing show only on button led, I don't know why, I guess I need to go back to Kickstarter and try the pins myself.
In yaml I have the following which I believe are the button leds:
```
output:
  - platform: libretiny_pwm
    id: output_led_1
    pin:
      number: P14
      inverted: true
  - platform: libretiny_pwm
    id: output_led_2
    pin:
      number: P26
      inverted: true
  - platform: libretiny_pwm
    id: output_led_3
    pin:
      number: P11
      inverted: true
light:
  - platform: monochromatic
    id: light_switch_1
    output: output_led_1
  - platform: monochromatic
    id: light_switch_2
    output: output_led_2
  - platform: monochromatic
    id: light_switch_3
    output: output_led_3
```

2. Kickstarter ESP turns off light switches after a period of time when the device is not connected to home assistant. How can I disable this?_

Cossid, 2023-12-09:
`monochromatic` may not work for all LEDs, you may want to try `binary` instead.

alecsandes, 2023-12-10:
Super! Thank you @Cossid! 
Do you know why the switch is turning off after a while if not connected to wifi/home assistant instance by chance? 
How can I prevent this?

Cossid, 2023-12-10:
Yes, ESPHome will reboot if HA hasn't connected for 15 minutes.  You can disable it by changing the value of `reboot_timeout` on the `api` component: https://esphome.io/components/api.html?highlight=reboot_timeout