libretiny-eu/upk2esphome #12: Tuya device firmware read from WR3 can't be analyzed in UPK2ESPHome

helgek, 2024-04-02:
Hi,

the firmware file for this humidifier device https://eurom.nl/en/product/climate-control/humidifiers/oasis-303-wifi/ is not readable in UPK2ESPHome. I never booted the device before reading the flash memory, so the file attached is original without any personal settings.

When I open the file in text editor I can see Tuya references and generally there's a lot of of readable text.
[Luftbefeuchter ebay ltchiptool_ambz_2024-04-02_00-45-27.zip](https://github.com/libretiny-eu/libretiny/files/14829553/Luftbefeuchter.ebay.ltchiptool_ambz_2024-04-02_00-45-27.zip)

Thanks,
Helge

helgek, 2024-04-03:
I think I've now understood better, thank you. So the WR3 (RTL8710B) on the board is only acting as WiFi interface and a different chip on one of the boards (this humidifier has several) would actually act as the controlling device. The WR3 has only TX and RX pins connected. I disassembled two more humidifiers and discovered WBR3 (RTL8720CF) chipsets.  One of these seemed to have more pins involved than TX and RX which might be then one with TuyaMCU + IO on the wifi, the other seemed to also only use TX and RX.

The WR3 based humidifier from which I now managed to read the flash memory is actually available in two versions, one with wifi and one without (https://eurom.nl/en/product/climate-control/humidifiers/oasis-303-wifi/ vs https://eurom.nl/en/product/climate-control/humidifiers/oasis-303/ ). So if I understood you right and the TuyaMCU is a separate chip besides the WR3 in my case then this would make even more sense that the WR3 only communicates through TX/RX. Assuming this is the case I wasn't able though to identify the TuyaMCU. The text on the other chips was not readable.

As a next step I will flash ESPHome as you suggested. These are pictures of the boards in the device:

1. WR3

[image]
[image]

2. 

[image]
[image]

3. Sensor

[image]
[image]

4. Control panel - display is based on LEDs 

[image]
[image]

Cossid, 2024-04-03:
All modules need 3 pins to power on, 3.3V, EN (CEN), and Ground.  Beyond that, any TuyaMCU needs an RX/TX, and anything else needs one or multiple GPIO.

Yes, the last image the chip labeled `U4` is an MCU.  They often don't have any markings on them.  The MCU handles all the input and certain pre-defined actions, so the device usually operates normally without the wifi unit entirely.

Here is the schema/dpid info that Tuya provides for the device.  With ESPHome and the `tuya:` module added, you should see some of these dpids advertised in the boot message, and you'll need to map them with tuya platform sensors and buttons.

```json
{
    "activeResponse": {
        "schema": [
            {
                "mode": "rw",
                "property": {
                    "type": "bool"
                },
                "id": 10,
                "type": "obj"
            },
            {
                "mode": "ro",
                "property": {
                    "range": [
                        "0",
                        "1"
                    ],
                    "type": "enum"
                },
                "id": 101,
                "type": "obj"
            },
            {
                "mode": "rw",
                "property": {
                    "range": [
                        "cancel",
                        "1",
                        "2",
                        "3"
                    ],
                    "type": "enum"
                },
                "id": 102,
                "type": "obj"
            },
            {
                "mode": "rw",
                "property": {
                    "type": "bool"
                },
                "id": 103,
                "type": "obj"
            },
            {
                "mode": "rw",
                "property": {
                    "type": "bool"
                },
                "id": 104,
                "type": "obj"
            },
            {
                "mode": "rw",
                "property": {
                    "range": [
                        "0",
                        "1",
                        "2",
                        "3",
                        "4",
                        "5",
[... 229 lines cut ...]
                            "typeSpec": {
                                "type": "value",
                                "max": 100,
                                "min": 0,
                                "scale": 0,
                                "step": 1,
                                "unit": ""
                            }
                        }
                    ]
                }
            ]
        }
    },
    "detailsResponse": {
        "category": "jsq",
        "category_name": "Humidifier",
        "icon": "smart/icon/ay15570525518720HELj/1559701980125d3648fa0.jpg",
        "model": "Oasis 303 Wi-Fi",
        "name": "Oasis 303 Wi-Fi",
        "product_id": "qcg6n0da0lnkuvkf",
        "product_name": "Oasis 303 Wi-Fi",
    },
}
```

helgek, 2024-04-03:
Thank you! I tried to flash the WR3 today, optimistic that it would easily boot again into flash mode. Unfortunately it didn't work out. I also tried with a bench power supply but I'm not sure if I connected everything correctly. Eventually I tried connecting the appliance to the 12V power supply to ensure a stable power supply from the appliance itself. This looked promising, in the terminal the WR3 displayed the Tuya boot data as as long as I would not try to reset it. Then at some point I made a wrong move with my equipment and accidentially touched with one of the test probes the power pins and damaged something on the board. So I'll have to give up for now and consider ordering another one later. There are two more humidifiers I've ordered and I want to check inside for which Tuya wifi chipset they use. Maybe I'm lucky and it's one of the Beken ones... 

Thank you again for your support and apologies that I've failed bringing it to a success! Maybe some other folks will find this thread and try their luck as well.

closed 2024-04-03 by helgek