libretiny-eu/upk2esphome #17: Missing entities for TOMZN TOB9-VAP

felipecrs, 2024-12-06:
Hi there,

First: thanks a lot for this project. It's tremendously helpful.

I noticed there it did not detect all the pins for TOMZN TOB9-VAP, even though they seem to be in the extracted data:

```jsonc
{
  "user_param_key": {
    "bt1_pin": 17, // button (NOT DETECTED)
	"led1_pin": 9, // red led (NOT DETECTED)
    "netled1_pin": 15, // blue status led
    "rl_off1_pin": 26, // bridge reverse (NOT DETECTED)
    "rl_on1_pin": 24, // bridge forward (NOT DETECTED)
  }
}
```

And this was the generated YAML:

```yaml
status_led:
  pin:
    number: P15
    inverted: true

uart:
  id: uart_bus
  tx_pin: TX1
  rx_pin: RX1
  baud_rate: 4800
  stop_bits: 1

sensor:
  - platform: bl0942
    uart_id: uart_bus
    current:
      name: BL0942 Current
    voltage:
      name: BL0942 Voltage
    power:
      name: BL0942 Power
      filters:
        multiply: -1
    energy:
      name: BL0942 Energy
    frequency:
      name: BL0942 Frequency
      accuracy_decimals: 2
```

This is the source file: [TOMZ-TOB9-VAP.source.json](https://github.com/user-attachments/files/18041569/TOMZ-TOB9-VAP.source.json)

This is the dump (remove .txt suffix): [ltchiptool_bk72xx_2024-12-05_20-25-23.bin.txt](https://github.com/user-attachments/files/18041785/ltchiptool_bk72xx_2024-12-05_20-25-23.bin.txt)

I built a device page in ESPHome for it: https://github.com/esphome/esphome-devices/pull/942

I just wanted to open this issue to document this finding.

kuba2k2, 2024-12-06:
This is most likely because the "H-bridge" switch is not supported yet. In fact, the ESPHome component that can support such a switch is not yet released, AFAIK.

The lack of the switch component makes it also not detect the button/LED.

felipecrs, 2024-12-06:
That's interesting. Thanks for the hint.

felipecrs, 2024-12-18:
H-Bridge has landed!

- https://github.com/esphome/esphome/pull/7421

kuba2k2, 2024-12-18:
Added to UPK2ESPHome - for now on the website only. Please check if the output is as intended (and if it works on device), by pasting the UPK data on the website.

felipecrs, 2024-12-18:
This was the output generated. It looks GREAT!

```yaml
output:
  - platform: libretiny_pwm
    id: output_led_1
    pin:
      number: P9
      inverted: true

light:
  - platform: monochromatic
    id: light_switch_1
    output: output_led_1

binary_sensor:
  - platform: gpio
    id: binary_switch_1
    pin:
      number: P17
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_1

switch:
  - platform: hbridge
    id: switch_1
    name: Relay 1
    on_pin: P24
    off_pin: P26
    pulse_length: 60ms
    wait_time: 30ms
    on_turn_on:
      - light.turn_on: light_switch_1
    on_turn_off:
      - light.turn_off: light_switch_1

status_led:
  pin:
    number: P15
    inverted: true

uart:
  id: uart_bus
  tx_pin: TX1
  rx_pin: RX1
  baud_rate: 4800
  stop_bits: 1

sensor:
  - platform: bl0942
    uart_id: uart_bus
    current:
      name: BL0942 Current
    voltage:
      name: BL0942 Voltage
    power:
      name: BL0942 Power
      filters:
        multiply: -1
    energy:
      name: BL0942 Energy
    frequency:
      name: BL0942 Frequency
      accuracy_decimals: 2
```

closed 2024-12-18 by felipecrs